DomCR/ACadSharp #578: DwgReader.Read incredibly slow on macOS

mvarie, 2025-03-11:
Using the same project and source file, `DwgReader.Read` takes a few seconds on Windows but _several minutes_ on macOS - regardless of whether the crc check is enabled.

Just in case, I’ve tested different files and IDEs while debugging, but the result is always the same.

I understand that non-Windows platforms might be unsupported, and I have no expectations in that regard. I’m just curious to know if this is a known issue, and if not, whether anyone has ideas on where to look for potential causes.

DomCR, 2025-03-11:
Hi @mvarie,

I've noticed that using the .NET Framework vs .NET have a different performance time being .NET faster, have you tested in both?

If that doesn't help, I think that I won't be able to resolve this issue because it seems a platform limitation.

mvarie, 2025-03-11:
@DomCR I’m using .NET 8 on both platforms—it’s literally the same project. However, my macOS platform is significantly faster than the Windows machine I’m using for this test, which is quite odd. I regularly develop .NET Core (C#) applications on both platforms and have never noticed such a difference. In fact, when working with managed .NET Core without Win32 dependencies, performance is typically smooth across the board.

I’m going to investigate further, but before diving in, I wanted to check if this is a known issue. If possible, I’d appreciate it if you could keep this open—I’d like to report back with my findings, whether I identify the cause or not.

closed 2026-10-23 by DomCR