DomCR/ACadSharp #580: Exception thrown when trying to read DWG file

Lukeor, 2025-03-13:
Hello,

I was recently given the attached DWG file. Apparently the DWG was created in DWG TrueView 2025 and was saved as a 2004 DWG version.

When trying to read in the file using ACadSharp I get the following exception:

'Exception of type 'System.Exception' was thrown.'

With the following stack trace:

   at ACadSharp.IO.DWG.DwgStreamReaderBase.ReadBitLong()
   at ACadSharp.IO.DWG.DwgClassesReader.Read()
   at ACadSharp.IO.DwgReader.Read()

I am using ACadSharp version 1.0.9.

[Problematic File.zip](https://github.com/user-attachments/files/19233883/Problematic.File.zip)

Please let me know if you require any further info.

Kind regards,
Luke

DomCR, 2025-03-14:
Hi @Lukeor,

I've looked at this issue and I managed to fix it, but by changing some code that in theory is documented.

Looking at the file with a hex editor looks really bloated with a lot of padding but it seems that is an official one, even Autocad doesn't complain so once I'm sure the fix is stable I release the package.

I'll also add more info in the exceptions for the stream readers.

Thanks for the report!

Lukeor, 2025-03-14:
Hi @DomCR 

That's great new, thanks very much for looking into so quickly.

Once a stable release is out I'm happy to give it a test on my end.

Thanks once again,

Luke.

closed 2025-03-17 by DomCR

DomCR, 2025-03-17:
I'll publish the version 1.0.11 isolated with this changes, if you have time, please check with your files and let me know if it solves the issue.

Lukeor, 2025-03-18:
@DomCR I can confirm that in the latest version (1.0.11) I am now able to successfully read in the given CAD file using ACadSharp.

Thanks very much for your quick fix and roll-out! It is greatly appreciated.

Luke