DomCR/ACadSharp #607: How do I use TextAlignment in TextEntity?

pswhall, 2025-04-09:
This is my code for the text alignment but is doesn't work.
Could you help me figure out what's wrong?

```
            TextEntity textEntity = new TextEntity();            
            textEntity.Height = height;
            textEntity.Rotation = rotation * Math.PI / 180.0;            
            textEntity.Value = text;  

            textEntity.AlignmentPoint = position ;
            textEntity.InsertPoint = position ;
            textEntity.Normal = CSMath.XYX.AxisY;

            switch (hAlign)
            {
                case TextRender.HorzAlign.Left: textEntity.HorizontalAlignment = ACadSharp.Entities.TextHorizontalAlignment.Left; break;
                case TextRender.HorzAlign.Right: textEntity.HorizontalAlignment = ACadSharp.Entities.TextHorizontalAlignment.Right; break;
                case TextRender.HorzAlign.Center: textEntity.HorizontalAlignment = ACadSharp.Entities.TextHorizontalAlignment.Center; break;
                default:
                    textEntity.HorizontalAlignment = ACadSharp.Entities.TextHorizontalAlignment.Left; break;
            }
            
            switch (vAlign)
            {
                case TextRender.VertAlign.Top: textEntity.VerticalAlignment = ACadSharp.Entities.TextVerticalAlignmentType.Top; break;
                case TextRender.VertAlign.Bottom: textEntity.VerticalAlignment = ACadSharp.Entities.TextVerticalAlignmentType.Bottom; break;
                case TextRender.VertAlign.Center: textEntity.VerticalAlignment = ACadSharp.Entities.TextVerticalAlignmentType.Middle; break;
                default:
                    textEntity.VerticalAlignment = ACadSharp.Entities.TextVerticalAlignmentType.Baseline; break;
            }
            
            btr.Entities.Add(textEntity);
```

DJGosnell, 2025-04-09:
What is not working?  The details are a little sparse.
Do all the text entities align in the same location?
Is it a vertical alignment issue or a horizontal alignment issue or both?
Does the text even show up at all?
Are you generating a DWG or DXF file?
Is the generated DXF/DWG corrupted?
What version of the project are you using?

pswhall, 2025-04-10:
Hi DJGosnell  thank for your reply,

I used version 1.1.6 and generated a DWG file.
I tried to align each text according to the options.
but all text is show up and located same alignment(horizontal left).
I think that horizontal alignment is not working.

DomCR, 2025-04-10:
Hi @pswhall,

I've linked this issue to the corresponding parent, I'm sorry to say that by now there are no updates on that matter, I'll try to take a look as soon as I can.