DomCR/ACadSharp #613: Reading a Spline created by ACadSharp

mme1950, 2025-04-13:
**The bug**
I created a `Spilne` entity using ACadSharp and added it into a document. It could be written with no problem.
When I read the `Spline` entity it fails when it reads the `KnotsTolerance` field.

**Analysis**
The written Scenario value is 2, but modified to 1 due to the `Flags1` value; thus the `KnotsTolerance` field is tried to be read.
The scenario value is modified since the `Flags1` property is `None`. The `Flags1` property should be set internally as can be seen in the `CadSplineTemplate.Build` method, when the `Spline` properties entered.

DomCR, 2025-04-23:
Hi @mme1950,

I'm having some issues trying to reproduce the issue, could you share the code that you are using to create the SPLine and the expected result?

Thanks!

mme1950, 2025-04-23:
```
        public Spline AddQSpline(double x1, double y1, double x5, double y5, double x2, double y2) {
            Spline spline = new Spline() {
                Layer = _layer,
                ControlPointTolerance = 1E-7,
                Degree = 3,
                FitTolerance = 1E-10,
                Flags = SplineFlags.None,
                KnotTolerance = 1E-7,
                LineType = LineType.ByLayer,
                LineWeight = ACadSharp.LineweightType.ByLayer,
                LinetypeScale = 1,
                Color = ACadSharp.Color.ByLayer
            };
            spline.FitPoints.Add(new CSMath.XYZ(x1, y1, 0));
            spline.FitPoints.Add(new CSMath.XYZ(x5, y5, 0));
            spline.FitPoints.Add(new CSMath.XYZ(x2, y2, 0));

            addToOwnerEntities(spline);  // adds the spline to a document's or block's entities.

            return spline;
        }
```

DomCR, 2025-04-23:
Now I see what you mean, opening a branch to fix it.

Thanks for the report!

mme1950, 2025-04-23:
Do you think, the solution is sufficient?

DomCR, 2025-04-23:
I think that the flags should removed and create them only during the writing process, there is no dxf code for them and it doesn't make much sense to store them in the class.

What do you think of my approach?

closed 2025-04-24 by DomCR

mme1950, 2025-05-10:
> Giving the values that you can add or modify, I'll set the Flags1 to public so it's user managed.

Hi @DomCR,
don't you think the Flags1 values are completely determined by the other properties of the Spline object. It would be convenient, when they were set internally. Now, of course, I can set them, but have to understand how. If we understand how we can do it internally.