DomCR/ACadSharp #619: The angle of rotation that the MText class has.

woodbook666, 2025-04-17:
Excuse my poor English.

The MText class holds the rotation angle in the Rotation property using group code 50, just like the TextEntity class. However, my understanding is that the MTEXT entity represents rotation using the direction vector defined by group codes 11, 21, and 31. Therefore, I believe it would be better to rename the variable slightly to avoid confusion. What do you think?

DJGosnell, 2025-04-17:
The `Rotation` property is the main value that stores the rotation of the `MText`, base point based depending upon the selected `AttachmentPoint`.  I have not seen a DWG utilize the `AlignmentPoint` to define the rotation of a `MText`.  I do know the `TextEntity.AlignmentPoint` to determine where to insert the text relative to the `IntertPoint`.

I don't think renaming these AutoCAD defined variables would be helpful as they come from the specification.

woodbook666, 2025-04-18:
Ah, it seems I had misunderstood the specification because I had only seen MTEXT entities where the drawing direction was defined by the AlignmentPoint. So, in other words, while a value can be present in the Rotation property, if a direction vector is defined via the AlignmentPoint, that takes precedence. If that's the case, then the current variables might be just fine as it is. Thank you for the quick response.

closed 2025-04-18 by woodbook666