DomCR/ACadSharp #622: When converting a DWG file to a DXF file, the Chinese characters are garbled.

tianxin8206, 2025-04-21:
**Describe the bug**
I read a normal DWG file and then converted it into a DXF file, but the Chinese content is displayed abnormally.

**To Reproduce**
My Code:
```C#
var filePath = @"D:\cad2.dwg";
var dxfFilePath = @"D:\cadfile2.dxf";

var cadDocument = DwgReader.Read(filePath);
cadDocument.Header.CodePage = "GB2312";
using var writer = new DxfWriter(dxfFilePath, cadDocument, false);
writer.Write();
```

**Screenshots**
normal:

[image]

abnormal:

[image]

---

**Release needed**

- [ Y ] Yes, using the Nuget package.
- [ ] No, using the github repo.

DomCR, 2025-04-23:
Hi @tianxin8206,

Could you provide the file in the example to run the code on my end?

Thanks!

tianxin8206, 2025-04-24:
[You can click on this link to download the file.](https://obs-mkwszc-web.obs.cn-north-4.myhuaweicloud.com/cad2.dwg)
@DomCR

DomCR, 2025-04-29:
The issue is related to the `WidthFactor` is marked as optional and is not written because there is no flag to set the `WriteOptional` to true.

I'll open a PR to fix it.

Thanks for the report!

tianxin8206, 2025-05-06:
@DomCR Excuse me, has this bug been fixed? I have already upgraded to version v1.1.19 and set DxfWriter.Configuration.WriteOptionalValues to true, but the converted content is still incorrect.

DomCR, 2025-05-07:
Hi @tianxin8206,

I guess that you are referring to the title: 
[image]

If you have made changes in the text and increasing it's length, you will need to recalculate the `WidthFactor` or add a line break so the text doesn't overflow.

I'm sorry but this feature is not integrated in this library because it requires some external ones that provide information about the font, size...

tianxin8206, 2025-05-07:
Hi @DomCR 
It's not just the title. As long as it's a conversion to Chinese characters, it will be incorrect. And I didn't perform any other operations on the file during the conversion.

DomCR, 2025-05-07:
Ok, sorry for not seeing it but It's hard to notice when you don't know Chinese.

Steps:
- You load the file and re-write it to dxf and some characters are not what they should in the output.
- You haven't changed anything in the file, not the text, version or added some entities.

Could you give me an specific example so I can focus on the difference between both files? like 2 images, 1 with the input file and other with the output, with an specific character to look at so I can see the difference.

Sorry for asking all that 😅

tianxin8206, 2025-05-07:
Hi @DomCR 
I have attached the download link for the original DWG file that needs to be converted in my previous reply. Regarding the output content, you can use the code in my first question to convert it into a DXF file, and then compare it with the original DWG file after opening it. If you have difficulty recognizing the Chinese content, you can locate the area within the red frame of the CAD drawing that I attached, and make a simple comparison with the content of the original DWG file. Alternatively, you can attach the converted content to the conversation, and I will make the judgment.