DomCR/ACadSharp #629: Some entities in generated DWG are invisible

mme1950, 2025-04-24:
I am creating a new CAD document by the following mechanisms
* clone a BlocksRecord from a source document.
* create a new BlockRecord filled with cloned Entities from a dynamic-block state from a source document.
* create a MText entitiy and add it to the documents entity list.

Problems:
* Dimension entities in a cloned block are not visible.
  When I edit the block in AutoCAD I see the Dimension. When I change the style and then reselect the original style and close the block editor, the Dimension becomes visible in the document.
* Multileader entities in a cloned block are not visible.
  When I edit the block in AutoCAD the MultiLeader is invisible. I can select all (Ctrl-A) including the MultiLeader. When I change the style and reselect reselect the original style the MultiLeader becomes visible, after closing the clock editor also in the document.
* A MText created using ACadSharp directly added to document's entities ist invisible.
  I can select the MText, when I try to edit it (context menu) it becomes visible.

Did anyone have the same or similar problems and found a solution?

mme1950, 2025-04-24:
Investigation
* Create an MText in AutoCAD
* Clone this MText using ACadSharp -> The cloed MText is visible in AutoCAD
* Create an MText using ACadSharp, set DrawingDirection = LeftToRight and TextHeight -> The Created MText is invisible in AutoCAD
   Editing the MText in AutoCAD makes the MText visible.
   The MText property grid dos not show any difference.
* Comparing the Properties of the MText created by AutoCAD and the MText created by ACadSharp shows a difference of the property AlignmentPoint (0,0,0).
* Set AlignmentPoint to (1,0,0) -> MText becomes visible in AutoCAD.

@DomCR Should we initialize DrawingDirection and AlignmentPoint?

mme1950, 2025-04-24:
Investigation Dimension
* Dimensions created by AutoCAD added to the document's entities are visible in AutoCAD. The Dimension's Block property is null.
* Dimensions that ere members of a cloned BlockRecords are not visible in AutoCAD. The Dimension's Block property is null.
* Edit the inserted BlockRecord in AutoCAD: change style and restore previous style of the Dimension, Then save the blockRecord. The Dimension becomes visible.
* Comparing all Properties exposed by ACadSharp of the Dimension before and after it was edited shows a difference; The Dimension's Block property is set.

DomCR, 2025-04-25:
Hi @mme1950,

**Dimensions**: this is a known issue, here is more info about it:
- https://github.com/DomCR/ACadSharp/discussions/556
- https://github.com/DomCR/ACadSharp/issues/557
- https://github.com/DomCR/ACadSharp/pull/559

**Multileader**: I think is a similar issue as the dimensions, the property may be `BlockContent`? haven't taken a look on this one.

**MText**: here is a discussion about this:
- https://github.com/DomCR/ACadSharp/discussions/620
I'll open a branch to change the defaults and make sure that it works.

Sorry for the delayed answer and thanks for the report.

mme1950, 2025-04-25:
Thanks a lot, I am very glad that there is an idea to solve the problem.

MultiLeader:
* As far I understood `BlockContent` optionally references a block to be dispayed instead of a text label. So this does not correspond to the `Dimension.Block` property.
* I tried to investigate the MultiLeader problem and found that the MultiLeader created in AutoCAD, opened and saved by ACadSharp and reopened in AutoCAD becomes invisible.
* It may be related to the MultiLeaderStyle, I have to look at it a little closer.

mme1950, 2025-05-11:
Hi @DomCR,

I am now using the current version after this merge.

I think there are tree issues:

1. The `Dimension.Block` is always null, when it is set:
```
		public BlockRecord Block
		{
			get { return this._block; }
			set
			{
				this._block = this.updateTable(this._block, this.Document?.BlockRecords);
			}
		}
```
should be changed to:
```
		public BlockRecord Block
		{
			get { return this._block; }
			set
			{
				this._block = this.updateTable(value, this.Document?.BlockRecords);
			}
		}
```

2. The Block property is not cloned:
I added this line to the `Clone()` method
```
			clone.Block = (BlockRecord)this.Block?.Clone();
```

3. The reference to the Block is not witten:

```
			//this._writer.HandleReference(DwgReferenceType.HardPointer, dimension.Block);
			this._writer.HandleReference(DwgReferenceType.HardPointer, null);

```
```
			this._writer.HandleReference(DwgReferenceType.HardPointer, dimension.Block);
			//this._writer.HandleReference(DwgReferenceType.HardPointer, null);
```

mme1950, 2025-05-11:
When I generate the DWG (Clone a block with Dimension objects) everything seems to be fine. The `BlockRecords` are in the document and the `Dimension.Block` property is properly set. But the dimensions are not visible in AutoCAD.

When I open the  block editor and save the block the dimensions become visible.

When I view the saved DWG new additonal block records were created ond related to the dimensions I thik with the same content.

mme1950, 2025-05-12:
Good Morning @DomCR,
I made a change to ensure that each BlockRecord related to a dimension has a unique "*D..." name.
Unfortionately I still do not see the Dimensions in AutoCAD.
When I open and save the BlockRecord in the block editor the dimensions become visble.
New BlockRecords are created and related to the Dimensions.

mme1950, 2025-05-12:
I found this, the `FirstCorner` is writte as `FourthCorner`:

```
		private void writeSolid(Solid solid)
		{
			//Thickness BT 39
			this._writer.WriteBitThickness(solid.Thickness);

			//Elevation BD ---Z for 10 - 13.
			this._writer.WriteBitDouble((double)solid.FirstCorner.Z);

			//1st corner 2RD 10
			this._writer.WriteRawDouble(solid.FirstCorner.X);
			this._writer.WriteRawDouble(solid.FirstCorner.Y);
			//2nd corner 2RD 11
			this._writer.WriteRawDouble(solid.SecondCorner.X);
			this._writer.WriteRawDouble(solid.SecondCorner.Y);
			//3rd corner 2RD 12
			this._writer.WriteRawDouble(solid.ThirdCorner.X);
			this._writer.WriteRawDouble(solid.ThirdCorner.Y);
			//4th corner 2RD 13
			this._writer.WriteRawDouble(solid.FirstCorner.X);
			this._writer.WriteRawDouble(solid.FirstCorner.Y);

			//Extrusion BE 210
			this._writer.WriteBitExtrusion(solid.Normal);
		}
```

mme1950, 2025-05-16:
Hi @DomCR,
let me summarize the results of my investigations regarding the visibility of dimensions:

Scenarios
(1) Create a simple DWG with a Dimension in model space in AutoCAD, open and save (no modification) in ACadSharp.
      --> The dimension is **invisble** in when the document is opened in AutoCAD.
(2) Use a DWG document containing a block with a dimension, clone the block into a new document.
      --> The dimension is **invisble** in when the document is opened in AutoCAD, other content in the block is visible.
(3) Create dimension in ACadSharp
      --> The dimension is **visible** in AutoCAD

* The Dimension created in ACadSharp is not associated with an anonymous Block and XDictionary property is null.
   But is suprisingly visible in AutoCAD.
* Dimensions created in AutoCAD seem always to be associated with an anonymous Block and a Dictionary referenced by the XDictionary property.
   Not visible.
* Setting the XDictionary property and the Block property to null does not help.
* The associated BlockRecord is now cloned and written by ACadSharp. But the dimension is still invisible.
* The XDictionary property points to a Dictionary with one entry **AcDbContextDataManager** pointing to another dictionary ACDB_ANNOTATIONSCALES with one entry pointing to an ACDB_ALDIMOBJECTCONTEXTDATA_CLASS object ("*A1") that cannot be resolved. This looks similar to the MultiLeader problem, that could be fixed by supporting the ACDB_MLEADEROBJECTCONTEXTDATA_CLASS.

Do you think these problems will be solved with #559?