DomCR/ACadSharp #639: Hi guys. I want to load raster images into CAD.What shall I do? Can you give me an example？

Armida220, 2025-04-27:


DomCR, 2025-04-29:
Hi @Armida220,

Here is an example of how to add a raster in a document:

```C#
ImageDefinition definition = new ImageDefinition();
definition.Size = new XY(1, 1);
definition.Name = "image";
definition.IsLoaded = true;

definition.FileName = "..\\..\\image.JPG"; //Path relative to the saved dwg/dxf, I recommend using a full path to avoid errors.

RasterImage raster = new RasterImage(definition);

raster.Flags = ImageDisplayFlags.ShowImage;

raster.ClipBoundaryVertices.Add(new XY(0, 0));
raster.ClipBoundaryVertices.Add(new XY(0, 1));
raster.ClipBoundaryVertices.Add(new XY(1, 1));
raster.ClipBoundaryVertices.Add(new XY(1, 0));

CadDocument document = new CadDocument();

document.Entities.Add(raster);
```

DomCR, 2025-05-07:
Hi @Armida220,

This is a known issue, I'm not sure why it happens, but the behavior of the image seems to be as expected.

I'll keep this issue up in case I find what's happening.

DomCR, 2025-05-07:
- https://github.com/DomCR/ACadSharp/issues/597