DomCR/ACadSharp #649: Can I read the template or specify a file, write it directly to the graphic entity, and save it as a file directly

jasnature, 2025-04-30:
I found that I can't create some graphic entities directly after reading the file. I can only clone the read file into the created caddocument after creating it with your caddocument, and then insert the entity.

example
```
//filePath use default 2007 template 
CadDocument doc;
using (DwgReader reader = new DwgReader(filePath))
{
    reader.Configuration.KeepUnknownNonGraphicalObjects = true;
    reader.Configuration.KeepUnknownEntities = true;
    reader.Configuration.ReadSummaryInfo = true;
    doc = reader.Read();
}

using (DwgWriter writer = new DwgWriter("c://12356.dwg", doc))
{
    //writer.Configuration.WriteAllHeaderVariables = true;
    writer.Configuration.WriteXData = true;
    writer.Configuration.WriteXRecords = true;

    writer.Write();
}
```
autocad throw error

> Unhandled Access Violation Reading 0x0004 Exception at 645718c9h

jasnature, 2025-04-30:
[Aus.zip](https://github.com/user-attachments/files/19970396/Aus.zip)

append file,I read this file directly and save it. AutoCAD can't open it either

DomCR, 2025-05-04:
Hi @jasnature,

Are you using the package? if so, which version?

jasnature, 2025-05-06:
> Hi [@jasnature](https://github.com/jasnature),
> 
> Are you using the package? if so, which version?

ac1018 , autocad 2004

> [Aus.zip](https://github.com/user-attachments/files/19970396/Aus.zip)
> 
> append file,I read this file directly and save it. AutoCAD can't open it either

Please use the above aus file to directly read and write the content, and then save as。

Then there is another question: how to disable the alternative style (dimoverride)

thks。。

DomCR, 2025-05-07:
I've found an issue regarding the `XRecords`, the file is fine as long as you deactivate the flag `WriteXRecords`.

Let me know if it fixes your issue.

> Then there is another question: how to disable the alternative style (dimoverride)

I'm investigating all related thing to dimensions and styles, I'm still figuring out how they work in the file, sorry but I'm not sure how to provide support for this right now.

closed 2025-05-08 by DomCR