DomCR/ACadSharp #651: Failed to read dxf file

pswhall, 2025-05-02:
When reading a dxf file, the message "The layer color cannot be ByLayer or ByBlock" is displayed.

1. I see a layer color setting code that checks as shown below. Is it absolutely necessary?
2. Is there a problematic value among the color values ​​of the layer in my dxf file?

ACadSharp/Tables/Layer.cs
```
/// <summary>
/// Color
/// </summary>
/// <remarks>
/// if the index is negative, layer is off
/// </remarks>
[DxfCodeValue(62, 420, 430)]
public Color Color
{
	get { return this._color; }
	set
	{
		if (value.IsByLayer || value.IsByBlock)
		{
			throw new ArgumentException("The layer color cannot be ByLayer or ByBlock", nameof(value));
		}

		this._color = value;
	}
}
```

this is dxf layer information

[image]

DomCR, 2025-05-04:
Hi @pswhall,

Can you provide an example of a file failing that way? I don't see any layer with a strange color.

> I see a layer color setting code that checks as shown below. Is it absolutely necessary?

The constrain is necessary to ensure a valid file output.

pswhall, 2025-05-07:
thanks for your reply.
 
It's hard to provide the file. Instead, I'll share a screen capture showing the debugging values ​​where the error occurs.

[image]

The Layer Color index value of the template CadObject read is 7, and the value read from _reader.ValueAsShort is 0.

Can you help me solve this problem?

DomCR, 2025-05-07:
Can you show the lines in the dxf where this fails? if is in ascii format. also, can you look for the version of the file?

pswhall, 2025-05-12:
version is AC1018
At the DxfTablesSectionReader.cs line 179,   the "this._reader.ValueAsString" value "14" is Fail.

pswhall, 2025-05-12:
At the DxfTablesSectionReader.cs

this._reader.Code is '5'

[image]

DomCR, 2025-05-12:
The code 5 stands for the `Handle` of the object, I think that's not the issue.

If you cannot provide the file, can you copy the Layer codes, or print an image? something like this:

```
  0
LAYER   // affected layer
  5
10
102
{ACAD_XDICTIONARY
360
E6
102
}
330
2
100
AcDbSymbolTableRecord
100
AcDbLayerTableRecord
  2
0
 70
     0
 62
     7
  6
Continuous
370
    -3
390
F
347
98
348
0
  0
LAYER   // next layer
  5
275
```

Or how did you generate the file? was it with Autocad or another software that I can use to get a similar one?

pswhall, 2025-05-13:
This is created by owned software.

999
DXF exporter 2013. LLC Alfa Lab. All rights reserved.
0
SECTION
2
HEADER
9
$ACADVER
1
AC1018
9
$HANDSEED
5
99
9
$DIMTXT
40
20
0
ENDSEC
0
SECTION
2
CLASSES
0
ENDSEC
0
SECTION
2
TABLES
0
TABLE
2
VPORT
5
1
330
0
100
AcDbSymbolTable
70
1
0
VPORT
5
11
330
1
100
AcDbSymbolTableRecord
100
AcDbViewportTableRecord
2
*Active
70
0
12
0
22
0
16
0
26
0
36
1
17
-4916.051
[... 3721 lines cut ...]
0
0
INSERT
5
98
100
AcDbEntity
8
S5DLegendLayer
67
1
100
AcDbBlockReference
2
S5DLegendRecord
10
1070
20
836
30
0
0
ENDSEC
0
EOF