DomCR/ACadSharp #661: onEntryNameChanged is never removed

mme1950, 2025-05-08:
**Description**
Cloning and renaming the cloned a `BlockRecord` changes the key of the original `BlockRecord` in the `CadDocument.BlockRecord` dictionary.

**To Reproduce**
* call Clone() at a `BlockRecord`
* set the name of the colned `BlockRecord`.
> The key of the entry of the original BlockRecord is change to the new name of the cloned object.

**Expected behavior**
Renaming the cloned BlockRecord must not change existing entries.

**Additional context**
* `item.OnNameChanged += this.onEntryNameChanged;` is called when the `BlockRecord` is added to the `BlockRecords` dictionary.
   `item.OnNameChanged -= this.onEntryNameChanged;` should be called, when the item is rmoved.
* Event handlers referring to the owner dictionary must be cleared in Clone.

DomCR, 2025-05-08:
Opening a PR, thanks for the report!

DomCR, 2025-05-08:
Hi @mme1950,

I think this issue was fixed:

- https://github.com/DomCR/ACadSharp/pull/650

Is still happening? if so, can you provide a code example?

Thanks!

mme1950, 2025-05-08:
Oh, sorry. This was quite recently. I have to update.

mme1950, 2025-05-08:
It works, perfect!

closed 2025-05-08 by mme1950