DomCR/ACadSharp #671: Polyline + Arc Explode issue

Vilo176, 2025-05-13:
Hi friends,

I want to walk through a Polyline do draw it.
Do do that, I use Polyline.Explode().

If the Polyline contains an "bulged" vertex, Explode() creates an Arc thanks to the "bulge" parameter.
Arc is valid, but when bulge is <0, Arc description does not follow the Polyline "path" anymore, it is inverted.

A quick/durty fix would be to store bulge into Arc, so that the final caller can take it into account to discretize Arc the right way.

DJGosnell, 2025-05-13:
I think that storing additional bulge data inside an exploded arc is the answer.  The explode method should be adapted for negative bulges.  @Vilo176 Is this something that you can work on?

Vilo176, 2025-05-14:
The issue is not straightforward.
To make it append, you have to change :
- Explode()
- Arc.PolygonalVertexes()
- CurveExtensions.PolygonalVertexes()

By the way, there are other issues with CurveExtensions.PolygonalVertexes() ...
I'll try to figure out the "proper" fixes, and share it.

closed 2026-10-23 by DomCR