DomCR/ACadSharp #674: Change Summary Info in DXF

sverduyn, 2025-05-26:
Hi,

I would like to change the CadSummaryInfo but it seems the DXFWriter does not save that info.

sverduyn, 2025-05-28:
Hi @DomCR,

The DXF file format definitely supports the properties. Both the standard and Custom properties are written to the DXF file. Se attached files as an example and also the screenshots below which was taken after opening the DXF file. I would be nice if you could include this functionality in a future update. We have tried in 2 other libraries and they both can create and update the DXF SummaryInfo.

[Test_2000.zip](https://github.com/user-attachments/files/20476600/Test_2000.zip)

[image]

[image]

Regards.

DomCR, 2025-05-28:
It seems that dxf stores the summary in a `XRecord`, inside the `RootDictionary`. I can check how to setup a feature that saves that information in the dxf automatically, in the mean time, you could read the dxf, look for those records  and re-create them into a new file.

What application are you using to see these values? is it native from windows?

sverduyn, 2025-05-28:
Autocad