DomCR/ACadSharp #683: Linetype appearance not copyed

thaiguer, 2025-06-14:
### Description
After copying a polyline from an existing DWG to a new one, the linetype appearence is not copied or not shown correctly in the new file.

### Expected behavior
I expect to have the same appearance on both files.

### Screenshots:
Existing File Linetype Manager:
[image]

Existing File Polyline Appearance:
[image]

New File Linetype Manager:
[image]

New File Polyline Appearance:
[image]

### Release needed:
- [ ] Yes, using the Nuget package.
- [X] No, using the github repo.

DomCR, 2025-06-16:
Hi @thaiguer,

I'm trying to replicate this issue but it seems to work fine on my end.

Can you provide a snipped of code of how are you moving/cloning the Polyline into the new file?

thaiguer, 2025-06-16:
> Hi [@thaiguer](https://github.com/thaiguer),
> 
> I'm trying to replicate this issue but it seems to work fine on my end.
> 
> Can you provide a snipped of code of how are you moving/cloning the Polyline into the new file?

Thats the code I'm using to copy the entities:
```
foreach (var entity in doc.Entities)
{
    if (entity.Layer.Name == layer.Name)
    {
        newDoc.Entities.Add((Entity)entity.Clone());
    }
}
```

DomCR, 2025-06-16:
Sorry but I still cannot replicate the issue, can you provide the file or a sample? 

I have a guess but I'm not sure if its the right one.

When you add an entity in a new document, if the document contains table records with the same name as the one you are adding, the record of the entity is replaced by the one existing in the document, so, the first entity that you are adding that adds that `LineType` may be deleting the segments, causing all the entities that use that type to be continuous. 

I would suggest to check when the `LineType` is added and see if it has the segments like the original document, or check if the `LineType` has Segments at all. In the second scenario, it may be an issue with the reader.