DomCR/ACadSharp #684: How to insert coordinate elevation points into a certain layer

kjz1997, 2025-06-15:
hello
I currently possess an elevation file in XYZ format. My process involves reading the DWG file, filtering points within the specified range based on the TK layer range defined in the DWG file, and subsequently inserting the GCD layer into the DWG file. Prior to insertion, entities within the GCD layer are deleted. Simultaneously, two attributes are assigned to the inserted elevation points: one as an integer and the other as a decimal, corresponding to the integer and decimal portions of the Z coordinate, respectively. Furthermore, color mappings are utilized for visualization based on attribute values, rather than employing text annotations for display. Below is my code. Although the generated DWG file successfully inserts points into the GCD layer, it exhibits significant differences from the original DWG file, with many functionalities remaining unimplemented. 01.dwg represents the original DWG file, while 1.dwg corresponds to the file generated via my code. Could you please advise which areas require modification?

`using ACadSharp;
using ACadSharp.Entities;
using ACadSharp.IO;
using ACadSharp.Tables;
using CSMath;
using System;
using System.Collections.Generic;
using System.IO;
using System.Linq;

namespace DepthPointProcessor
{
  public class DwgProcessor
  {

[file.zip](https://github.com/user-attachments/files/20744047/file.zip)

    public static void ProcessDwgWithAttributedDepth(string inputDwgPath, string depthFilePath, string outputDwgPath)
    {
      Console.WriteLine($"开始处理DWG文件: {inputDwgPath}");
      CadDocument doc = null;
      try
      {
     
        doc = ReadDwgFile(inputDwgPath);

     
        BoundingBox tkBounds = GetTkLayerBounds(doc);

   
        List<XYZ> rawDepthPoints = ReadXYZDepthFile(depthFilePath);

  
        List<XYZ> filteredDepthPoints = FilterPointsByBounds(rawDepthPoints, tkBounds);
        Console.WriteLine($"在TK图层边界内筛选出 {filteredDepthPoints.Count} 个水深点。");

        ClearLayerEntities(doc, "GCD");

        
        AddAttributedDepthPoints(doc, filteredDepthPoints);

 
        SaveDwgFile(doc, outputDwgPath);
        Console.WriteLine($"处理完成，结果保存至: {outputDwgPath}");
      }
      catch (Exception ex)
      {
        Console.WriteLine($"处理过程中发生错误: {ex.Message}");
      }
      finally
      {
 
      }
    }

    private static CadDocument ReadDwgFile(string filePath)
    {
      using (var reader = new DwgReader(filePath))
      {
        Console.WriteLine($"正在读取DWG文件: {filePath}");
        return reader.Read();
      }
    }

    
    private static BoundingBox GetTkLayerBounds(CadDocument doc)
    {
[... 343 lines cut ...]
  {
    public static void LogConsoleNotification(object sender, NotificationEventArgs e)
    {
      Console.WriteLine($"[ACADSHARP NOTIFICATION] {e.Message}");
    }
  }

  class Program
  {
    static void Main(string[] args)
    {
   
      string inputDwg = @"F:\cad\02.dwg";  
      string depthData = @"F:\cad\0530_北段_8m.xyz"; 
      string outputDwg = @"1.dwg";

      DwgProcessor.ProcessDwgWithAttributedDepth(inputDwg, depthData, outputDwg);

      
      Console.ReadKey();
    }

  }
}
`

DomCR, 2025-06-18:
Hi @kjz1997,

I've read both files, it seems that they contain REGION entities, which are still not supported by the library. I think that's the main difference that you see.

If that's not the difference I would need you to be more specific.

kjz1997, 2025-06-19:
> DomCR

I have solved it thanks

iPhone

DomCR, 2025-06-19:
Hi @kjz1997,

Perfect!

One question before closing the issue, I see that the attributes are duplicated in the second screenshot, is that intended or is caused by some unhandled issue in `ACadSharp`?

closed 2026-10-23 by DomCR