DomCR/ACadSharp #696: Writing a SortEntitiesTable in ACadSharp creates corrupted DWG

mme1950, 2025-06-25:
I created a simple DWG with a Multileader and Wipeout, the Wipeout pushed to background.
The DWG can be read with ACadSharp. A SortEntitiesTable is associated with the model space.

When the DWG a SortEntitiesTable is saved by ACadSharp it is corrupted:
* In AutoCAD an error occurs when the DWG is read, The SortEntitiesTable has no effect.
* In ACadSharp an error occurs when the Sorter entries of the SortEntitiesTable are read. The values of the Sorter.Handle is invalid.

> This is the code to read a Sorter entry:
> ```
> 				//Sort handle(numentries of these, CODE 0, i.e.part of the main bit stream, not of the handle bit stream!).
> 				//The sort handle does not have to point to an entity (but it can).
> 				//This is just the handle used for determining the drawing order of the entity specified by the entity handle in the handle bit stream.
> 				//When the sortentstable doesn’t have a
> 				//mapping from entity handle to sort handle, then the entity’s own handle is used for sorting.
> 				ulong sortHandle = this._objectReader.HandleReference();
> 				ulong entityHandle = this.handleReference();
> ```
> these are corresponding lines to write a Sorter entry:
> ```
> 				this._writer.HandleReference(item.Handle);
> 				this._writer.HandleReference(DwgReferenceType.SoftPointer, item.Entity);
> ```
> Both calls run through the same code. I do not understand the details. According to the comment the two handles should be written in different sections.

DomCR, 2025-06-26:
Hi @mme1950,

I've recently merged this simple PR:

-https://github.com/DomCR/ACadSharp/pull/654

I thought it worked but I can see it stills create corrupted files, I'll rollback the changes and open a new one for this issue.

mme1950, 2025-06-26:
I do not know whether I understand the concept. 
Does
				ulong sortHandle = this._objectReader.HandleReference();
read a handle value from the main stream and
				ulong entityHandle = this.handleReference();
from the handles section?

So an "objectWriter" should be used to write the handle value to the main stream.

DomCR, 2025-06-28:
You are right! even the comment says it:

`//Sort handle(numentries of these, CODE 0, i.e.part of the main bit stream, not of the handle bit stream!).`

closed 2025-07-08 by DomCR