DomCR/ACadSharp #706: MTEXT rotation

sigster, 2025-07-10:
Hi

How cad I set Mtext rotation

Regards
Sigster

thaiguer, 2025-07-11:
The property Rotation of the MText object is read only.
But there is a method called ApplyRotation. I think that is it.

```
var mTextObject = new MText();
mTextObject.ApplyRotation(new CSMath.XYZ(0, 0, 0), 0.7854);
```

sigster, 2025-07-11:
I am reading shapefile -27.587405 is the angel off the text
ApplyRotation don't rotate the text in the right way

`    Dim ROT As Double
    ROT = -27.587405

    Dim mtext As New MText()
    mtext.ApplyRotation(New CSMath.XYZ(10, 10, 0), ROT * Math.PI / 180.0)
    mtext.InsertPoint = New CSMath.XYZ(10, 10, 0)
    mtext.Value = "HELLO I'm a rotated MTEXT"
    mtext.Height = 50
    mtext.Layer = doc.Layers(6)
    doc.Entities.Add(mtext)`

DomCR, 2025-07-12:
Hi @sigster,

To change the rotation of an `MText` you only need to change the `AlignmentPoint` to the direction that you want.

`ApplyRotation` works with an axis, so if you want to rotate in the same plane as the model you should use the axis Z instead of (10, 10), also we found some issues regarding this method and the text entities, by now, I'll recommend to change the `AlignmentPoint`.

Hope this helps!

sigster, 2025-07-12:
Thanks

How do AlignmentPoint  work if I want to rotate the text -30

Any change it is problem with mtext.Column.ColumnType = ColumnType.DynamicColumns
I can no set Defined Width and Defined height

[image]

DomCR, 2025-07-14:
Use this as an example to create the vector:

`mtext.AlignmentPoint = new XYZ(Math.Cos(MathHelper.DegToRad(-30)), Math.Sin(MathHelper.DegToRad(-30)), 0);`

I haven't work with columns, I'll have to take a look and come back to you.