DomCR/ACadSharp #727: Support for BOM

domeitzinger, 2025-08-06:
Thanks for this great library! 

The issue I am facing is that I have to read the BOM defined in a DWG file (as well as the balloon references for the parts). This would seem to require the implementation of the following types (from what I can tell by the logs): `ACMPARTLIST`, `ACMPARTREF`, `ACMBALLOON`

I would really appreciate it if this was possible.

DomCR, 2025-08-06:
Hi @domeitzinger,

Thanks for the support!

First, it can be done, but, the entities that you describe are not documented anywhere, nor dxf or dwg, which means that they have to be investigated manually, for dxf is easy, just an ascii file gives away a lot of information for dwg is a little bit more tricky.

Do you have a file that I can use where this objects are present?

domeitzinger, 2025-08-06:
Sure, I could send you a DWG (as well as a PDF export for reference) per mail. Should I use the mail address in your profile?