DomCR/ACadSharp #731: Add a header property for system variable CMLEADERSTYLE

mme1950, 2025-08-11:
The current MultiLeaderStyle is represented by the system variable CMLEADERSTYLE.
But there is no header property to get and set it.

Properties to be added (similar to current DimensionStyle):
```
CurrentMultiLeaderStyle { get; }
CurrentMultiLeaderStyleName { get; set; }
```

CMLEADERSTYLE is not written to DXF as $CMLEADERSTYLE.
The value is found in the `AcDbVariableDictionary` (entry of the root dictionary).

mme1950, 2025-08-15:
Hi @DomCR,

the two new properties will simply be linked to the respective dictionary entry.
There is no need of an explicit call in the reader, and there will be no `CMLEADERSTYLE` property in DwgHeaderHandlesCollection.

What do you think about this appoach?

DomCR, 2025-08-15:
Yep! sounds good!

I've been looking and this entry is not explicitly saved for DWG, only in dxf as a system var, so your approach will work.