DomCR/ACadSharp #741: PDFunderlay Clone

ChandrasekarML, 2025-08-19:
I am currently working with AcadSharp and would like to know how to properly clone a PdfUnderlay object from an existing dwg file. Could you please provide me with a sample or example code for this?

This is a sample code that we tried:

    var originalDef = targetDocument.PdfDefinitions.OfType<PdfUnderlayDefinition>()
     .FirstOrDefault(pd => pd.Name == "NCR_DAPSAURA_System Configuration_R01-K5D_20250513 - 1");

    if (originalDef != null)
    {
        var definition = new PdfUnderlayDefinition();
        definition.Page = "1";
        definition.File = originalDef.File;

        definition.Name = $"{definition.File} {definition.Page}";
  
        PdfUnderlay raster = new PdfUnderlay(definition);
        raster.InsertPoint = new XYZ(40, 140, 0);
        raster.ClipBoundaryVertices.Add(new XY(10, 10));
        raster.ClipBoundaryVertices.Add(new XY(100, 110));

        newDocument.Entities.Add(raster);
    }
}

DomCR, 2025-08-19:
Hi @ChandrasekarML,

I see that the Clone() method is not completed, I'll open a branch to fix it.

Thanks for the report!

closed 2025-08-19 by DomCR

ChandrasekarML, 2025-08-26:
Please provide sample code.