DomCR/ACadSharp #757: DXF export creates empty ENTITIES section (ACadSharp 3.1.12, sample_AC1014.dwg)

ShylockGao, 2025-08-29:
---

**Describe the bug**
When saving a DWG file to DXF using ACadSharp, the generated DXF file cannot be opened in AutoCAD. The DXF `ENTITIES` section is empty, and AutoCAD reports that the file may be corrupted or damaged. The ACadSharp version is 3.1.12, and the DWG file used is the sample sample_AC1014.dwg from GitHub.

**To Reproduce**
Steps to reproduce the behavior:

```csharp
using ACadSharp;
using ACadSharp.IO;

// Read DWG
var reader = new DwgReader("test.dwg");
var doc = reader.Read();

// Save as DXF
using (var writer = new DxfWriter("output.dxf", doc))
{
    writer.Write();
}
```

**Expected behavior**
The exported DXF file should contain all entities from the DWG (lines, circles, polylines, etc.) and be openable in AutoCAD or other CAD software.

**Screenshots**
*If applicable, add a screenshot of the DXF opened in a text editor showing empty `ENTITIES`.*

**Additional context**

* The DXF file is created with a valid header, but the `ENTITIES` section is completely empty.
* AutoCAD and other viewers refuse to open the DXF.
* The source DWG contains simple geometry (lines, polylines).
* Example DWG file is attached for reproduction.

---

**Release needed**

* [x] Yes, using the Nuget package.
* [ ] No, using the github repo.

---

DomCR, 2025-08-30:
Having problems with the investigation PR being automatically merged.

ShylockGao, 2025-09-01:
Thanks for the clarification! I'll try with AC1032 as a workaround and wait for the PR. Appreciate your support. 🙏

DomCR, 2025-09-01:
All the tasks done to solve this issue are linked in the PR above.

There still is an issue with the Shape entities due an added complexity, these entities need to read an SHX file to work properly, to bypass this issue, there is an option for the writers to ignore this type of entities and ensure a valid file.