DomCR/ACadSharp #768: CadHeader.CurrentMultiLeaderStyle property removed

mme1950, 2025-08-31:
TheCad.Heder.CurrentMultiLeaderStyle property was added with PR https://github.com/DomCR/ACadSharp/pull/736 but was removed with https://github.com/DomCR/ACadSharp/commit/4e78ea46bc50925d1ada0100e87e061bb2f652e2.

mme1950, 2025-08-31:
I just found it, thanks.
Sorry to have disturbed you on Sunday.

DomCR, 2025-08-31:
No worries I'm working on the project 🤣

mme1950, 2025-08-31:
The GetCurrent/SetCurrent appoach is OK. It provides a unique mechanism to access "Current" system variables. However, the header properties are more intuitive. 
But it actually is no header system variable.
From the perspective of a developer using ACadSharp it is not clear why there is a difference between the current `DimensionStyle` and the current `MulitiLeaderStyle`.