DomCR/ACadSharp #801: Hatch.BoundaryPath.Ellipse.CounterClockWise is always false

mme1950, 2025-09-21:
The `Hatch.BoundaryPath.Ellipse.CounterClockWise` property is never read.

The `Hatch.BoundaryPath.Ellipse` class has two properties for `DxfCodeValue(73)`:
* `CounterClockWise`
* `IsCounterclockwise`

Only `IsCounterclockwise` is set by the reader, `CounterClockWise` is always false.

I suggest to remove `IsCounterclockwise`, because
* boolean properties starting with `Is` are unusual,
* `IsCounterclockwise` should rather be  `IsCounterClockWise`
* The `Hatch.BoundaryPath.Arc` class has the property `CounterClockWise` only.

@DomCR please assign this issue to me. I will open a PR.

mme1950, 2025-09-22:
OK, I still would propose to keep `CounterClockWise` because it is consistent with `Hatch.BoundaryPath.Arc`.
If we rename `IsCounterclockwise` to `IsCounterClockWise` it is a breaking change anyhow, so we can remove it as well.

DomCR, 2025-09-24:
O may... yes please, add this change, thanks!